AllenNeuralDynamics/Aind.Behavior.VrForaging #56: Odor solenoid kept opened after end of session

tiffanyona, 2023-09-19:
Obsevations were:
1. Animal was doing something weird, licking and stopping outside of the odor site even though he appeared to be understanding that logic (672107-20230919T123120). 
2. Box smelled like banana and air seemed to be coming out of the odor tube. 
3. Odor solenoid 1 appear hot upon touch. 

Fix: Everything went back to normal after start and stopping the workflow once again. 

**Hypothesis 1**. Bonsai closed while the odor was on. **Add a stop button to finish the session that ensures that everything is closed.** 
**Hypothesis 2**: Animal 107 was running really fast. Perhaps he went so fast that the solenoid was stuck? Observe if the situation repeats after fixing hypothesis 1.

bruno-f-cruz, 2023-09-20:
Almost for sure te first. Should be fixed with #57

bruno-f-cruz, 2023-09-20:
Closed via [23bf090](https://github.com/AllenNeuralDynamics/aind-vr-foraging/commit/23bf0901a6e2dd4146e4940c419d58fcef951070)

closed 2023-09-20 by bruno-f-cruz