AllenNeuralDynamics/Aind.Behavior.VrForaging #202: PR issues

tiffanyona, 2024-01-28:
Issues observed when testing the PR:

- Odor valves don't open (no red light when there should be on the olfactometer). Raw data looks like there is only one READ command and no WRITEs even though there are several EndValve triggers.
- Patch identity and Site-Intersite-Interpatch plots are not working in the main visualizer.
- Change default water given by ForceReward to 3.

tiffanyona, 2024-01-28:
- Water calibration seems to not be reversed yet. When I changed the fit so it looks like the calibration (opening times bottom and values recovered/iteration number), the valves opens too much in the task. 

For instance: 
Before I inputted :
Slope: 0.02
Intercept: -0.02
[image]

Now I input: 
Slope: 43.05
Intercept: 1.22
[image]

bruno-f-cruz, 2024-01-28:
Re: Water. This is what is currently going on in Bonsai

`delay(s) = water(a.u.) * CalibrationSlope + CalibrationIntercept`

I will change it to:
`delay(s) = (water - offset) / slope` so you can use the values from WR2.
I must have gotten confused when changing the model, sorry!

bruno-f-cruz, 2024-01-28:
These should have all been fixed via 5be7aba3bb9a64f52f64aeb0ba5885e3c3ac5e8d

closed 2024-02-11 by bruno-f-cruz