AllenNeuralDynamics/Aind.Behavior.VrForaging #229: Test rig assembly

tiffanyona, 2024-04-10:


nathan-mars, 2024-04-18:
Great, I will try to help with the items as I can. The easy stuff I can grab anytime you want - motor controller, motor controller cables, Ethernet switch (not sure you will actually need this for a single computer?), lickspout, any screws, and the IR led power supply. I will check my inventory on IR lenses. 

We're waiting on a delivery of more IR LEDs and photodiode boards. 

The harder items that I have to build are the stage and lickspout holder. If you have time to help build a stage that could expedite it, otherwise it may take a bit. Lickspout holder is a quick 3d print.

cindypoo, 2024-04-18:
We can also prioritize setting up the olfactometer first -- if we have a complete system for that, it will help troubleshooting and calibration of olfactometer a lot.

travisjramirez, 2024-04-18:
I have parts for the olfactometer and will work on today. I can give you all an update tomorrow.

travisjramirez, 2024-04-19:
Here are pictures of the olfactometer assembly. Not sure if it works, but we shall see!

[image]
[image]

cindypoo, 2024-04-24:
Add enclosure to test rig.

cindypoo, 2024-04-26:
@travisjramirez, @nathan-mars 
I talked to @bruno-f-cruz: can we prioritize connecting these 3 items to the computer for the test rig by EOD on Monday? 
- 1 HARP behavior board
- 1 camera
- VR screen assembly (3 screens, mounted and connected)

please let me know if you need help with any of these @bruno-f-cruz is also available to help connect these things up on Monday if needed.

travisjramirez, 2024-04-26:
On it. Will post here when completed

tiffanyona, 2024-05-07:
Hi! I had a look at the olfactometer assembly the other day because Helio asked me to. The organization of the tubing looks good (only missing the tube from the end valves to the animal) but the connections are all really loose and can be easily disconnected. After screwing in, you need to tug them to make sure that they don't disconnect.

closed 2024-06-06 by bruno-f-cruz