AllenNeuralDynamics/Aind.Behavior.VrForaging #244: [Z:\scratch\vr-foraging\data\716458\20240422T091200] Register__StopCameras.bin sends two stop messages when session ended.

NgD-Huy, 2024-05-01:
Per Bruno's request,

When verifying that camera metadata.csv matched that of harp data (ie., *SideBodyCamera\Metadata.csv* and *Register__Camera1Frame.bin*), I noticed that *Register__StopCameras.bin* sent two stop messages when the session ended. 

[image]

**Comparing the CSV file and the bin file of the full session**
Since the timestamps match 1:1, the (blue and green) lines overlap perfectly.
[image]

**Zoomed-in view to better show Register__StartCameras.bin and  Register__StopCameras.bin**
The difference between two stop signals is a 1000th of a millisecond so the difference cannot be seen on the chart.
[image]

bruno-f-cruz, 2024-05-01:
Can you also add the data of the CamTrigger dataframe including the first 10 and last 10 frames? thanks!

NgD-Huy, 2024-05-01:
> Can you also add the data of the CamTrigger dataframe including the first 10 and last 10 frames? thanks!

#### **Register__Camera1Frame.bin**
[image]

#### **SideBodyCamera\Metadata.csv**
[image]

bruno-f-cruz, 2024-05-01:
Please match the timestamp axis. Your previous post zeros all timestamps whereas your last one doesn't. Thanks!

NgD-Huy, 2024-05-01:
> Please match the timestamp axis. Your previous post zeros all timestamps whereas your last one doesn't. Additionally, try to include text instead of images. Copy-paste to code blocks limited by "```" works fine. Thanks!

Updated comment

bruno-f-cruz, 2024-05-01:
As far as I can tell all data looks good.
First frame and last frame arrive in between the start and stop pwm. I will check the workflow but i suspect I am just sending two messages to stop the cameras (one when i stop the logging and a second one when bonsai closes all the hardware). This has no effects on the actual data so not that urgent to fix.

NgD-Huy, 2024-05-02:
### **Register__StartCameras.bin**

<table border="1" class="dataframe">
  <thead>
    <tr style="text-align: right;">
      <th></th>
      <th>0</th>
      <th>MessageType</th>
    </tr>
    <tr>
      <th>Time</th>
      <th></th>
      <th></th>
    </tr>
  </thead>
  <tbody>
    <tr>
      <th>1809641.9474879999</th>
      <td>0</td>
      <td>READ</td>
    </tr>
    <tr>
      <th>1809641.9545600000</th>
      <td>1</td>
      <td>WRITE</td>
    </tr>
  </tbody>
</table>
</div>

### **Register__StopCameras.bin**
<table border="1" class="dataframe">
  <thead>
    <tr style="text-align: right;">
      <th></th>
      <th>0</th>
      <th>MessageType</th>
    </tr>
    <tr>
      <th>Time</th>
      <th></th>
      <th></th>
    </tr>
  </thead>
  <tbody>
    <tr>
      <th>1809641.9495039999</th>
      <td>1</td>
      <td>READ</td>
    </tr>
    <tr>
      <th>1812810.4904960000</th>
      <td>1</td>
      <td>WRITE</td>
    </tr>
    <tr>
      <th>1812810.4938560000</th>
      <td>1</td>
      <td>EVENT</td>
    </tr>
  </tbody>
</table>
</div>

bruno-f-cruz, 2024-05-02:
There appears to be an undocumented EVENT type message from this register. I am tracking the issue here: https://github.com/harp-tech/device.behavior/issues/18

Otherwise everything looks good with our workflow

closed 2024-05-07 by bruno-f-cruz