AllenNeuralDynamics/Aind.Behavior.VrForaging #246: Harp timestamps bellow expected threshold

NgD-Huy, 2024-05-08:
# Session: 715869\20240507T080801
## fps: 120 - time between frames: 8.33ms

For this session, two harp timestamps were spaced ~6 ms from each other instead of 8.33 ms. This does not seem to be a dropped frame issue as (1) camera timestamps and frame diff show no anomaly and (2) no compensations were observed, meaning the diff goes back to the expected 8.33 ms. This is consistent across FaceCamera/Metadata.csv, SideBodyCamera/Metadata.csv, and Camera0Frame.bin as timestamps matched exactly.

[image]
Check that the timestamps match exactly. Delta = 0 means no difference between the timestamp of these three files.

[image]
The time between frames dipped twice in Harp Timestamp (blue) while Camera timestamp (orange) and frame diff are normal

[image]
Harp Timestamp only view

### Dip location
Index: 187420
Timestamp: [2992896.995136, 2992897.001824, 2992897.010176] 
Diff: [0.006688 0.008352]

Index: 341023
Timestamp: [2994176.998432, 2994177.00464,  2994177.01296 ]
Diff: [0.006208 0.00832 ]

bruno-f-cruz, 2024-05-08:
Tracking here https://github.com/harp-tech/device.behavior/issues/20

bruno-f-cruz, 2026-05-16:
Closed via #476

closed 2026-05-16 by bruno-f-cruz