AllenNeuralDynamics/Aind.Behavior.VrForaging #294: Add calibration lookup table for treadmill brake

bruno-f-cruz, 2024-08-28:
This will likely require changing the current rig schema and adding a first citizen class to the aind behavior services
There are two options to keep this data:
- lookup table
- fitted function parameters

@Poofjunior can you give me an example of what data we should be keeping, along with domains of the data?

Once we decide this format, we also need to refactor the bonsai code to calibrate the online manipulation

Poofjunior, 2024-08-29:
The calibration data I'm collecting is:
* input: current (12-bit but upscaled to 0 - 65535)
* output: measured torque (12-bit, not upscaled, with 2048 being the "zero torque" output.)

We want a table that stores:
* input: desired current
* output: actual current to produce desired current

bruno-f-cruz, 2024-09-05:
Implemented in #289

closed 2024-11-15 by bruno-f-cruz