AllenNeuralDynamics/Aind.Behavior.VrForaging #296: Treadmill reports erroneous write command

tiffanyona, 2024-09-03:
[image]

How to reproduce:
1. Run a session in bonsai.
2. Click end session. 
3. Close bonsai. 
4. Open a new client with the launcher and prepare a new session. 

After bonsai opens, it will show this error. The way to fix it is power-cycling the board. It happens every time, even if nothing was touched on the hardware side. 

The only box that displays this behavior is 4A, that treadmill had to be flashed at a later time than all the others since the original treadmill was faulty. Possibly different firmware than all the other boards.

bruno-f-cruz, 2024-09-04:
Can you add what the SOP is to start a session? 

Is it always:
1. Connect treadmill
2. Open bonsai
3. Start session

Or does this order change?

NgD-Huy, 2024-09-09:
> Can you add what the SOP is to start a session?
> 
> Is it always:
> 
> 1. Connect treadmill
> 2. Open bonsai
> 3. Start session
> 
> Or does this order change?

Order for me is 
1. Open bonsai --> flush water + audio testing
2. Connect treadmill
3. Start session

tiffanyona, 2024-09-09:
Huy is correct, we need to open a session in bonsai to prepare all the settings to run the session. Only afterwards we load the mouse to the treadmill and then put the treadmill inside the box.

bruno-f-cruz, 2024-09-10:
Here are the edge cases I am concerned about:

1. Is the treadmill never there when bonsai starts? (Never attempt to set friction to 0) If not, can I always clear the limit flag anyway?
2. Will the treadmill ALWAYS be added BEFORE clicking START (exception will be swallowed/handled until start)?
3. Will the treadmill NEVER be touched AFTER clicking START (exception will surface)?
4. Does the treadmill need to be set to 0 friction immediately after the animal wheel is connected OR can we wait until the session starts? (excepting will be swallowed/handled until start)

cindypoo, 2024-09-11:
1. No
(We don't need 4)
2. Yes. 
3. Yes. It will never be touched.
4. We don't need this

PR #300 (2024-09-17): Defer assertation of treadmill state until experiment starts
Closes #296
closed 2024-09-17 by bruno-f-cruz