AllenNeuralDynamics/Aind.Behavior.VrForaging #297: Choice cue and water delivered outside of odor site

tiffanyona, 2024-09-06:
I am detecting instances where the choice cue is delivered following a stop OUTSIDE of an odor site. 

I am seeing this in the data after I run the analysis (there is the possibility that there is an error on the analysis, but nothing obvious so far and I double-checked). I also got the impression that I saw it happen online, while running the mice. 

Generally it happens more frequently at the end of sessions. 

This is an example in time
[image]

And the same session in space
[image]

I have the suspicion that I might be something about the grace distance threshold or something of sorts.

The specific session I am showing here is 754579_20240905T131137

bruno-f-cruz, 2024-09-06:
I think the bug here is something else. Are you sure data is correctly aligned? Because the animal did not stop in the first site...

bruno-f-cruz, 2024-09-06:
I can see a situation where the animal stops in the inter-site and this can occur, I am actually surprised we did not observe this earlier.

tiffanyona, 2024-09-06:
I think I had seen it sporadically before but really rarely and I thought it was something off in my analysis. Now I am seeing it really regularly with this batch.

closed 2024-09-21 by bruno-f-cruz