AllenNeuralDynamics/Aind.Behavior.VrForaging #303: Friction is not being set from task logic

bruno-f-cruz, 2024-09-14:
I suspect a bug with the Interpolator as there are several 0s logged as harp messages. This is weird as DistinctUntilChanged is applied prior to the interpolator.

bruno-f-cruz, 2024-09-17:
Does not appear to be a bug with the interpolator implementation either. A test (F(x), x) shows that everything is working as expected:
```
(0, 0)
(6553, 0.1)
(13107, 0.2)
(19660, 0.3)
(26214, 0.4)
(32767, 0.5)
(39321, 0.6)
(45874, 0.7)
(52428, 0.8)
(58981, 0.9)
```

bruno-f-cruz, 2024-09-20:
@tiffanyona can you add the task logic json and commit necessary to reproduce this bug?

tiffanyona, 2024-09-20:
Yes! 
Commit: 9c6cc34
Task logic:
[test_friction.json](https://github.com/user-attachments/files/17080204/test_friction.json)

I used rig 5A for these tests, in case you want to do the same

bruno-f-cruz, 2024-09-20:
Sorry I meant the rig schema with the calibration.

bruno-f-cruz, 2024-09-20:
Turns out the 0.4.x version had a bug in the interpolator that has been fixed in the proposed 0.5 release. I have patched it and made a new 0.4.3 release with this fix!

PR #305 (2024-09-20): Fix treadmill interpolator
Closes #303
closed 2024-09-20 by bruno-f-cruz