AllenNeuralDynamics/Aind.Behavior.VrForaging #304: Error "Camera dropped frame detected during pre-session" when using 500fps

tiffanyona, 2024-09-17:
Right after running bonsai and before starting the experiment, the error message in the title pops up and stops bonsai. 
The rig has two cameras and they are set to 8bit depth, manually in Spinview. 
If the bit depth is set to 10bits, it won't throw this error, it will let us run the experiment, but it will constantly pop up the warning about bad alignment. 

Rig schema: [4C_video.json](https://github.com/user-attachments/files/17032918/4C_video.json)
Task logic: [stageC_v1.json](https://github.com/user-attachments/files/17032923/stageC_v1.json)

bruno-f-cruz, 2024-09-17:
Can you additionally add the model of the camera and the hash of the commit you are running this from?

bruno-f-cruz, 2024-09-20:
This is expected behavior. 

The camera has a max frame rate of 226 fps. The error that you are getting is telling you that something is wrong with your rig configuration and one of the cameras cannot keep. To prevent experimenters from acquiring corrupt data, I throw an error if this is the case and the experiment is not allowed to continue.
https://softwareservices.flir.com/BFS-U3-16S2/latest/Model/spec.html

If you want the two cameras at two different frame rates just instantiate two independent controllers instead of a single one.

closed 2024-09-21 by bruno-f-cruz