AllenNeuralDynamics/Aind.Behavior.VrForaging #307: Metadata that needs fixing

saskiad, 2024-10-17:
- [ ] Add `project_name` to data_description

In Session:

- [ ] "Lick sensor left" and "Lick sensor right" are not daq_names for your rig and thus can't be listed as under `daq_names`. 
- [ ] Camera names don't match the names in the rig. They must match. There is a utility function you can use to make sure that the names align between rig and session.
- [ ] Reward delivery cannot be null. How are you delivery rewards without it?
- [ ] Your `session_end_time` is before the end of one of your data streams. The end time should be the last time of all data streams and stimulus epochs.

Stimulus Epoch
- [ ] Missing `stimulus_device_name`
- [ ] These output parameters and task parameters duplicate almost the entire metadata schema. Can we please please please not do this? Very seriously, this is going to make our DocDB painful and it ultimately undermines having the schema in the first place.
- [ ] eg. calibrations should be in the calibration field
- [ ] frame rate should be in the device (also, check the version of the schema you're using. The cameras in your device class have a `max_frame_rate` which was changed a while ago)
- [ ] Your `frame_num` item specifically is blocking upload to DocDB. You can't have fields with ".csv" in the name. This information should live in the quality_control class (which was only recently added to the schema, but that's where it should live)

bruno-f-cruz, 2024-10-17:
Adding @saskiad feedback from teams here since it appears the previous report was from a different task asset:

- [ ] Need experimenter name (either real name or an anonymized version, your team decides how you want to do it)
- [ ] need session_end_time - this is surprising to me
- [ ] Need iacuc protocol. There much be an iacuc protocol for these experiments
- [ ] Olfactometer Channel Config does not belong in the calibration, it lives in the stimulus 
- [ ] The ***WHEEL*** information also doesn’t belong in the calibration but in the mouse platform device. (Also, this is a wheel. There is a separate treadmill class. This really needs to be consistent and clear - and this applies to the parameters things as well. Interchanging wheel/disc/treadmill is going to breed confusion across our data if we let it slide)
- [ ] Watervalve calibration would benefit from some unit definitions
- [ ] Some of the things on your daq list are not daq devices. They might be connected to a daq, but they themselves are not daq. Eg screen. Manipulator.
- [ ] Given the stimulus modalities you list, I expect your list of stimulus devices to include an olfactometer, monitor, speaker, reward delivery,
- [ ] Given the presence of a speaker, I expect speaker config

bruno-f-cruz, 2024-11-01:
Closed by #309

closed 2024-11-01 by bruno-f-cruz