AllenNeuralDynamics/Aind.Behavior.VrForaging #313: Issues while testing 0.5.0 release

tiffanyona, 2024-11-13:
1. The VR display lags a bit, it's not as fluid as before. 
2. The name of the mouse and task at the bottom is cropped (independently of the maximization of the window
3. The naming convention for the data folders changed, right?
4. end experiment moves the spout vertically (Z) instead of backwards (y1 and y2)
5. I am not sure what should I expect after I press end experiment. It basically stops the workflow but hangs everything else with no message in terminal information of the state? I only tried with the launcher, I will try it in VS, maybe it is a bit more informative.

tiffanyona, 2024-11-13:
6. Lookup table not behaving as intented. 
[test_PR.json](https://github.com/user-attachments/files/17738002/test_PR.json)

bruno-f-cruz, 2024-11-14:
1. TBD
2. Fixed in 1f59004f4f73f3f5732692a69352fe88c94a583f
3. Already documented in the PR (#289 and in [Services docs](https://allenneuraldynamics.github.io/Aind.Behavior.Services/articles/core/dataset_structure.html). If something doesn't match, it should be considered a bug and reported.
4. Fixed in 9f3c31f06f98ff718d0aabef2cff06e20acb765b

closed 2024-11-15 by bruno-f-cruz